adafruit/Adafruit_MPR121 #9: Compatibility with ATSAMD21

Axagthoth, 2016-04-19:
I'm aware that the Readme of this library states it doesn't work with ATSAMD21 cores, however I find myself in a use-case scenario where that would be precisely the ideal setup. Is there something intrinsic with the chip that impedes it functioning together with these processors, or could anything be done to fix the library and add compatibility for ATSAMD21 cores? (Specifically an Arduino MKR1000)

ladyada, 2016-04-19:
it might now because recently the repeated start i2c code was merged in! but we havent specifically tried it. its likely to work, if it does for u plz reply and we'll update!

Axagthoth, 2016-04-19:
Well, it turns out it does! (Though I have no idea what you mean by "the repeated start I2C code"). I got a quirky behaviour on the first upload, but now I've triple checked and it seems to work flawlessly, even with multiple different devices on the I2C bus. Great news! BTW Adafruit rocks!

ladyada, 2016-04-19:
yay

closed 2016-04-19 by ladyada

Axagthoth, 2016-04-19:
Dang I forgot. I changed the example code because _BV(i) isn't defined for the ATSAMD21. Just changed it to (1<<i).