adafruit/Adafruit_MPR121 #15: Ported version for Trinket

estranged42, 2018-02-17:
I think I've managed to port this library over to use TinyWireM so that it works on the Trinket 5v.

https://github.com/estranged42/Adafruit_TinyMPR121

I've updated the example sketch in that repo and it seems to work fine. 

Many thanks to Rik Lesniak who ported the Adafruit_TinyCAP1188 library and provided a great model to copy for this port.

ladyada, 2018-02-17:
hiya, Trinket supports Wire, so there's no reason you'd need TinyWireM?

estranged42, 2018-02-18:
I thought that was the case, but some discussions on the Discord channel led me down the TinyWireM path.

However during that work I discovered that the real culprit seems to be the while() lines on 108 and 116: 
`while (Wire.requestFrom(_i2caddr, 1) != 1);`

For some reason these cause the Trinket to hang and never progress, preventing the `begin()` method from ever exiting.

I changed these to:
`Wire.requestFrom(_i2caddr, 1);`
and
`Wire.requestFrom(_i2caddr, 2);`
while leaving the original Wire library in place, and those two changes got this library working on the Trinket!

I'm not sure what the waiting for those requests to equal 1 and 2 do, but they don't seem to work as intended on the Tinket.

hoffmannjan, 2019-06-06:
the newest version should have this fixed, please test It and reopen if problem persist. 👍

closed 2019-06-06 by hoffmannjan