adafruit/Adafruit_MPR121 #20: setThresholds does not work as expected

caternuson, 2018-10-17:
- Arduino board:  UNO Rev 3
- Arduino IDE version (found in Arduino -> About Arduino menu): 1.8.7
- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): Add something like this as last line in `setup()`
```cpp
  cap.setThresholds(0xFF, 0x00);
```
The MPR121 requires being in STOP mode to write to registers. 
[image]
This is being done in the constructor, but not else where. So calling `setThresholds()` has no effect.

fritsjan, 2019-05-15:
fixed, can be closed now

hoffmannjan, 2019-05-15:
👍

closed 2019-05-15 by hoffmannjan