adafruit/Adafruit_MPR121 #21: MPR121 library can work for Arduino Due, if...

jrokken, 2018-10-25:
- Arduino board:  Arduino Due (ATSAM3X8E)

- Arduino IDE version: v1.8.7, but I mostly use Eclipse and Sloeber 4.3.1 plugin.

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): Arduino Due did not work with the Adafruit MPR121 library and it also states that on your project page.  Just open the libraries example (MPR121test) and download to the Due and it outputs that the MPR121 is not found

I was able to track down why the Due does not work.  The Wire library does not handle the I2C restart bit correctly on the due even though there was a parameter added endTransmission() to specifically handle that.  The ATSAM3x8e I2C peripheral will automatically generate the Write Cmd containing the internal address followed by the restart when the Read Cmd address size is non-zero.  The MPR121 library was always using address length of 0 and then performing the Write Cmd manually and then the Read Cmd.  Because of that, the Write Cmd still generated a STOP bit on the Due.  The datasheet for the MPR121 does not like seeing that STOP bit and it then returns 0's for the Read Cmds because it tosses the internal address away after seeing the STOP.  It only accepts the RepeatStart between the Write and Read transmissions.

I do not know if the following change breaks other Arduinos, but it did fix the Arduino Due.  If other Arduinos would properly handle the endTransmission() SendStop parameter, they likely would still work.  I just don't have any to try.  The fix is to remove the Write and replace requestFrom() with one of the overloaded functions that handle the internal address.

here's the change:
@@ -176,10 +176,7 @@
  *  @returns    the 8 bit value that was read.
  ****************************************************************************************/
 uint8_t Adafruit_MPR121::readRegister8(uint8_t reg) {
-    Wire.beginTransmission(_i2caddr);
-    Wire.write(reg);
-    Wire.endTransmission(false);
-    Wire.requestFrom(_i2caddr, 1);
+    Wire.requestFrom(_i2caddr, 1, reg, 1, 0);
     if (Wire.available() < 1)
       return 0;
     return (Wire.read());

jrokken, 2018-10-26:
Ok sorry, the + and -'s were turned into bullet points.  The first 4 are -'s and last one is +.

I missed readRegister16() before though I am not sure where it would be used on MPR121 since I don't think there are any 16 bit addresses.  The same change could be made there with the exception of the size of 2 instead of 1 as the second parameter.

jrokken, 2018-10-26:
I located a Mega2560 and tested the above changes.  No problems found.

Grippy98, 2019-02-19:
Just writing in to confirm your change fixes it working on the Due. You have to use Pins 20 and 21 not SDA1 and SCL1 by default but once moved it works great.

Thank you!

Haven't tested if this breaks it with any other board but for this it's great.

thietbihitech, 2019-04-24:
i am find librali arduino due for proteus to [make packing machine ](http://thietbihitech.vn/detail/may-dong-goi-tra-tui-loc-ca-phe-tui-loc-kieu-tui-dang-banh-tron-2102.html) kindly hepl me. thanks.

hoffmannjan, 2019-06-06:
please make PR with this changes so we can test It on other boards too

ladyada, 2020-11-12:
refactored library, so all i2c is done in busio library, should be much better for due and other platforms now
https://github.com/adafruit/Adafruit_MPR121/pull/34

closed 2020-11-12 by ladyada

Nobaiha, 2021-02-17:
Hi can someone please help me? I am very confused on what to change and where exactly... 
Wire.h file doesn't have anything remotely close to the above changed code..