adafruit/Adafruit_MPR121 #35: MPR121 library does not work with Arduino Due although all the libraries are up to date

yurikadoi, 2021-01-15:
- Arduino board:  Arduino Due

- Arduino IDE version (found in Arduino -> About Arduino menu): arduino 1.8.13

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 
I have all the libraries up to date but when I tried to run MPRtest, it says "MPR121 not found, check wiring?".
I made sure about the port (using programming port).
My arduino Due is fine with other programs that does not have MPR 121 part.
MPR 121's ground is connected to Arduino Due ground.
MPR 121's Vin is connected to Arduino Due 5V.
MPR 121's SDA is connected to Arduino Due SDA1.
MPR 121's SCL is connected to Arduino Due SCL1.
(I also tried Due's 20 and 21 pins for SDA and SCL)
I did not have pull-up resistors initially. But because it did not work, I tried pull-up resistors of 4.7k ohm but the result was the same.
My library versions are up-to-date:
arduino MPR121: ver 1.1.0
arduino busio: ver 1.7.1

I saw this old post (https://github.com/adafruit/Adafruit_MPR121/issues/21) and thought it is resolved but it is not.

ladyada, 2021-01-15:
if it still does not work, then it will never work. Due i2c was always very bad.

caternuson, 2022-07-12:
Appears to be DUE specific.

It'll show up in an I2C scan at least:
[image]

Running MPR121 test on same DUE:
[image]

It's getting back `0` for the `MPR121_CONFIG2` (0x5D) register read test for some reason. Scope trace:
[image]

Testing with QT Py M0, with some added serial prints in `begin()`:
[image]
works as expected.

Scope trace from M0 showing expected `0x24` read back from `MPR121_CONFIG2` (0x5D):
[image]

caternuson, 2022-07-12:
Looks like the MPR121 wants a repeated start for the single byte register read:
[image]

The DUE isn't doing that, despite being asked to :(
[image]

Closing since it appears to be an issue with the DUE Arduino core. Looks like there's at least one open issue for it here:
https://github.com/arduino/ArduinoCore-sam/issues/70

closed 2022-07-12 by caternuson