adafruit/Adafruit_MPR121 #39: MPR121 weak touch detection

xbavnu, 2022-02-17:
Hello,

I am having problems with the touch sensitivity of the MPR121. 
Even when I set the threshold value very low `cap.setThreshholds(2, 1);` the detection of the touch button is very weak. 
When I touch it directly on the PCB, it works okay, but as soon as I add anything on top of the PCB, the touch won't be recognized anymore. Even with a 1.5mm thick piece of acrylic it won't detect the touch anymore. With other touch sensors (like the TTP223) I can easily place a 6mm sheet of acrylic on top of the sensor and the touch will still be detected. 

First I thought the touch area was too small, but I tested it with circular touch areas with various sizes (from 8mm all the way up to 25mm) and the touch detection is pretty much the same with all of them.

Is there a way to further increase the sensitivity? Or is it normal, that the MPR121 only works directly on the PCB? But in which usecase would you touch directly on the PCB instead of a surface above it?

xbavnu, 2022-02-17:
Yes, I am. I am using the black version which has the connectors on the sides.

ladyada, 2022-02-17:
ok you can run the second half the test example code with cap.filteredData to see the raw data, and use larger cpacitive touch pads. if you need more help please post in the forums with photos :)

xbavnu, 2022-02-23:
Thank you for the response. I did a lot more testing and also tried using the raw touch data, unfortunately this also did not help. Even with the raw data I can barely see a touch detection with a 1mm piece of acrylic on top of the touch pad.

I designed a custom PCB with touch pads as well as the MPR121 on it. The length from the touch pad to the MPR121 is max. 150mm, but even with the pads close to the MPR121 there is no difference. On the picture you can also see the various sizes of touch pads I tested. I even tried removing the silkscreen and adding extra solder on top of it. But still, the touch sensitivity is very weak and I can not reliably detect anything with more than 0.5mm distance to the pad.

[image]

Is there any other way to increase the sensitivity? Changing a capacitor or resistor on the Adafruit MPR121 Board or something like this?

ladyada, 2022-02-23:
not that we see, you can look for MPR121 app notes and datasheets for advice!

xbavnu, 2022-02-23:
Thanks, I will have another look on the datasheet.
So is this the normal behaviour, that the touch sensors with the MPR121 are so weak, that they can not even go through 1mm of acrylic?

ladyada, 2022-02-23:
we havent tried it so we dont know for sure! we have always used bare metal pads

tahunus, 2022-07-01:
> Thanks, I will have another look on the datasheet. So is this the normal behaviour, that the touch sensors with the MPR121 are so weak, that they can not even go through 1mm of acrylic?

I've been playing with AFE & FCR Configuration Registers (see [https://www.nxp.com/docs/en/application-note/AN3890.pdf](url)) and can increase sensitivity to go through self-adhesive vinyl.  Could not make it work with acrylic. Just be careful not to overstress the units with too much charge (see [https://www.nxp.com/docs/en/application-note/AN3889.pdf](url)).  Both CONFIG parameters can be found in MPR121.cpp under the _Adafruit_MPR121::begin_ definition.

caternuson, 2022-07-12:
@tahunus  Those app notes look useful. Thanks. 

The `setThreshholds()` method only alters the touch/release threshold registers (`0x41`-`0x5A`).

There are no dedicated methods for changing the AFE & FCR registers.

**AFE Configruation Register** = `0x5C`
[image]
* **FFI** = First Filter Iterations – The first filter iterations field selects the number of samples taken as input to the first level of filtering.
* **CDC** = Charge Discharge Current – The Charge Discharge Current field selects the supply current to be used when charging and discharging an electrode.

**Filter Configuration Register** = `0x5D`
[image]
* **CDT** = Charge Discharge Time – The Charge Discharge Time field selects the amount of time an electrode charges and discharges.
* **SFI** = Second Filter Iterations – The Second Filter Iterations field selects the number of samples taken for the second level filter.
*  **ESI** = Electrode Sample Interval – The Electrode Sample Interval field selects the period between samples used for the second level of filtering.

However, the class's `writeRegister()` method is public. So can use that to alter those registers as needed from your sketch code. Something like this, somewhere after calling `cap.begin()` in your sketch's `setup()`:
```cpp
                                  // replace x's with 1's and 0's as desired
cap.writeRegister(MPR121_CONFIG1, 0bxxxxxxxx);  // AFE register (FFI and CDC)
cap.writeRegister(MPR121_CONFIG2, 0bxxxxxxxx);  // FCR register (CDT, SFI, and ESI)
```
That way the library code itself does need to be altered. The sketch code settings will override the hardwired values in `begin()`.

@xbavnu Try using the information in the app notes along with the methods above for tuning the registers for your setup and see if that helps. If you're not familiar with setting registers, the byte literal syntax shown above may make is easiest to set each specific bit as desired. For example, to set the FFI bits to `11` which sets samples taken to 34, and CDC to `000001` for 1uA:
```cpp
cap.writeRegister(MPR121_CONFIG1, 0b11000001);  // AFE register (FFI and CDC)
```

If having control of these two registers seems useful, we can look at adding some proper getter/setter methods and predefined enum settings to make it easier.

tahunus, 2022-07-17:
@caternuson consider that setThreshholds (two h's) has been deprecated and setThresholds (one h) is the one active.

Also: if you're looking at adding additional methods, may I suggest to consider the Autoconfig feature?  It does work quite nicely!  I've been tinkering with it and just wrote a [quick guide](https://github.com/tahunus/MPR121_Calibration)  of my use case that might be of benefit for others.