adafruit/Adafruit_MPR121 #42: Autoconfig function

wystewart, 2025-01-12:
recommend to  add a function to enable autoconfig or to add an optional parameter to begin to enable autoconfig.  The later being the preferred method.

Present method requires modification of code, which breaks well updating the code to a new version in arduinoide or platformio.

thank you.

wystewart, 2025-01-12:
Suggest code changes:

.h file
  bool begin(uint8_t i2caddr = MPR121_I2CADDR_DEFAULT, TwoWire *theWire = &Wire,
             uint8_t touchThreshold = MPR121_TOUCH_THRESHOLD_DEFAULT,
             uint8_t releaseThreshold = MPR121_RELEASE_THRESHOLD_DEFAULT,
             boolean autoconfig = false);

.c file
delete
// uncomment to use autoconfig !
// #define AUTOCONFIG // use autoconfig (Yes it works pretty well!)

modify
bool Adafruit_MPR121::begin(uint8_t i2caddr, TwoWire *theWire,
                            uint8_t touchThreshold, uint8_t releaseThreshold,
                            bool autoconfig) {

if (autoconfig)
{
  writeRegister(MPR121_AUTOCONFIG0, 0x0B);

  // correct values for Vdd = 3.3V
  writeRegister(MPR121_UPLIMIT, 200);     // ((Vdd - 0.7)/Vdd) * 256
  writeRegister(MPR121_TARGETLIMIT, 180); // UPLIMIT * 0.9
  writeRegister(MPR121_LOWLIMIT, 130);    // UPLIMIT * 0.65
}

s-light, 2025-06-11:
please @wystewart test my pr.
i think i incorporated the changes you wished in a nice way...